dannerph/homeassistant-keba #4: Add ConfigFlow tests

dannerph, 2022-01-16:


Taraman17, 2025-07-26:
I could work on this.
However before I start, I like to know, if you plan to integrate the changes back into the core integration again.
If so, the config_flow would be a good point to start, since before changing anything HA requires integrations to have one.

Since you can't bring over the whole integration in one PR, but have to split it, the config_flow would be a good starting point.
And I found writing tests is a lot easier if you directly do it in core.

So if you want to integrate into core and agree, I would start to port the config_flow over and add the tests there.
Else I can also first add the tests here.

skolima, 2026-08-27:
There's a working suite for this on a fork now — https://github.com/skolima/homeassistant-keba — built on top of #47, at 100% coverage of config_flow.py and 95% overall. I've left a longer comment on #47 about how to proceed, since the work builds directly on it.