dannerph/homeassistant-keba #5: Keba stoppped working

chani666, 2022-08-14:
After one of the updates in the last 2 months, the Keba P30 integration stopped working. I get these errors:

2022-08-14 12:19:53.817 DEBUG (MainThread) [keba_kecontact.keba_protocol] Asyncio UDP connection setup complete.
2022-08-14 12:19:53.831 DEBUG (MainThread) [keba_kecontact.keba_protocol] Send report 1
2022-08-14 12:19:53.944 DEBUG (MainThread) [keba_kecontact.keba_protocol] Send report 2
2022-08-14 12:19:54.130 DEBUG (MainThread) [keba_kecontact.keba_protocol] Send report 3
2022-08-14 12:19:54.260 ERROR (MainThread) [homeassistant.components.keba] Could not find a charging station at 192.168.1.18
2022-08-14 12:19:54.262 ERROR (MainThread) [homeassistant.setup] Setup failed for keba: Integration failed to initialize.

I tried to debug into this with the remote debugger. But I have no clue, I think the exception happens outside of the integration and has something to do with the upd connection. When I use an upd client and sent the reports to the Keba it responds just normal. But HA doesnt receive anything, and probably also doesnt sent the report commands.

This just worked perfectly until around June. Maybe with 2022.6 it still worked. Only thing I changed with my HA was installing MariaDB and adjusting the recorder. Any idea why this would stop working?

dannerph, 2022-08-14:
Do you have one keba charging station or multiples?
Which installation type (plain, docker, hassos, ...) are you using?

chani666, 2022-08-14:
I have only one, and it runs on hassos:

chani666, 2022-08-20:
I backuped the whole configuration and installed it on a fresh system. Now it works again. I guess I close it as not reproducible.

closed 2022-08-20 by chani666