dannerph/homeassistant-keba #6: After Charge Stop with button.keba_p30_disable a later charging with a re plug of the Cable and RFID Auth

skl283, 2022-08-19:
Hi @dannerph 

i have the following issue - i use a OCPP Backend and authorize with an RFID Card. Everything works perfect, but i only have an issue with the folowing workflow. I use an OCPP Backend and an RFID Card to authorize.

1. Stop Charging with button.keba_p30_disable
2. Unplug the Car, drive around
3. Hold my RFID in front of the Keba
4. Replug the car
5. Charge doesn't start, because of the Status i set with button.keba_p30_disable

If i press "button.keba_p30_enable" - before i put the RFID Card in Front of the Keba it works as expected. My Workaround ist to  create an automation which presses button.keba_p30_enable after unpluggig the car. I think, this is something which has to be done in Code to not leave the box in an "unknown"status.

dannerph, 2022-08-19:
Hi @skl283,
the described behaviour seems to be exactly how it should happen as far as I have understood it. The disable (udp: ena 0) always needs an enable (ena 1) before a charging process could start. The state should be ena 0 then, which is not unknown and this functionality might be required for other use cases, e.g. emergency stops, blocking all charging processes, ... However, I understand your concern.

Is it required to stop the charging process with the disable service? With my BMW i3, I just unlock the car an the charging process gets interupted from the car. Then I can unplug, drive around and charge afterwards without issues.

skl283, 2022-08-20:
Thank you for your quick Answer @dannerph!

I use button.keba_p30_disable / button.keba_p30_enable to use the pv excess similar to you post at your guide. If excess is too low, i stop the charging process and start it again, if the excess is higher.

If the pv excess is too low and i must drive my car the wallbox is disabled. The charging process won't start. I want to control the start and stop process by an automation/manual but i want to to control it everytime i plug/replug the car. 

Do i misuse the button.keba_p30_disable button? Is there a better way to just stop _this_ session and not effect the next session?

thank you
sash

dannerph, 2022-08-21:
The best option for you should be the "currtime 0 1" command, which according to the manual sets ena 0 with a delay of 1 second. After unplugging the car, it should reset the ena state.
You can use this command via the set_current service when you specify a delay > 0.

skl283, 2022-08-22:
Thank you! I'll try this, have just modified my Automatation to 
`service: keba.set_current
data:
  current: 0
  delay: 1`
and will try to get this working. :-)

closed 2022-08-22 by skl283