dannerph/homeassistant-keba #7: Deprecation Warnings

spali, 2022-11-06:
Seems to be still working even on 2022.11. But HA may drop the support anytime soon.

```
Entity None (<class 'custom_components.keba.number.KebaNumber'>) is using deprecated NumberEntity features which will be unsupported from Home Assistant Core 2022.10, please report it to the custom integration author.
custom_components.keba.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
custom_components.keba.number::KebaNumber is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
```

dannerph, 2023-03-05:
fixed in 2.4.0

spali, 2023-03-06:
Thanks @dannerph 

On `2023.3.1` and integration version `2.4.0`

One Warning less, but still getting the following:
```
WARNING (MainThread) [homeassistant.components.number] custom_components.keba.number::KebaNumber is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
WARNING (MainThread) [homeassistant.components.number] Entity None (<class 'custom_components.keba.number.KebaNumber'>) is using deprecated NumberEntity features which will be unsupported from Home Assistant Core 2022.10, please report it to the custom integration author.
```

And the following pops up since a some week of HA versions (can't remember in which exact version it has started):
```
WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.keba1_maximum_current_system_2 (<class 'custom_components.keba.sensor.KebaSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['mA', 'A']; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

Let me know if you want a new issue.

dannerph, 2023-03-06:
I just checked it: in my productive setup I get the warning as well, in my dev env (Home Assistant Core dev branch), there were no warnings yesterday. I will have a look at it in the evening.

It seems I have missed a sensor to convert to the new unit system

dannerph, 2023-03-06:
Aha, it appears I simply forgot to commit the number.py file 🙈 fixed in 2.4.1 now

closed 2023-03-06 by dannerph

spali, 2023-03-06:
Now I got:
```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant.config_entries
Source: custom_components/keba/__init__.py:86
Integration: Keba Charging Station
First occurred: 19:32:24 (1 occurrences)
Last logged: 19:32:24

Error setting up entry Keba for keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/keba/__init__.py", line 86, in async_setup_entry
    wallbox.set_failsafe(fs_mode, fs_timeout, fs_fallback, fs_persist)
TypeError: Wallbox.set_failsafe() takes from 1 to 4 positional arguments but 5 were given
```
😊

dannerph, 2023-03-06:
me as well, again not in my dev environment ...

dannerph, 2023-03-06:
finally fixed in 2.4.2 for me. Required a restart so that the correct library is installed

spali, 2023-03-06:
What you mean with restart? I do always install with HACS followed by a HA restart.
Tried two restarts, but now can't get rid of:
```
Logger: homeassistant.components.notify
Source: custom_components/keba/notify.py:39
Integration: Benachrichtigung (documentation, issues)
First occurred: 20:08:04 (1 occurrences)
Last logged: 20:08:04

Error setting up platform keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 89, in async_setup_platform
    notify_service = await platform.async_get_service(
  File "/config/custom_components/keba/notify.py", line 31, in async_get_service
    return KebaNotificationService(keba)
  File "/config/custom_components/keba/notify.py", line 39, in __init__
    self.targets: dict[str, Wallbox] = {}
AttributeError: can't set attribute 'targets'

```

dannerph, 2023-03-06:
Yes normal restart after HACS install.

I have not yet tackled the notify issue, please refer to #8

spali, 2023-03-06:
Thanks, will follow the mentioned issue. Just realized that except notify (which I don't use) the rest seems to work.
Appreciate you work 👍