dannerph/homeassistant-keba #8: Notify service stopped working in 2023.3

binzhou, 2023-03-02:
Error during setup:
```
Error setting up platform keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 89, in async_setup_platform
    notify_service = await platform.async_get_service(
  File "/config/custom_components/keba/notify.py", line 31, in async_get_service
    return KebaNotificationService(keba)
  File "/config/custom_components/keba/notify.py", line 39, in __init__
    self.targets: dict[str, Wallbox] = {}
AttributeError: can't set attribute 'targets'
```

dannerph, 2023-03-05:
I need to test it in the upcoming days, hopefully I can reproduce using my emulator, as my charging station does not have a display.

slovenec88, 2023-03-06:
@binzhou did you try with the version 2.4.0? I have Keba P30X and P30C, both have displays, I'm on 2023.3.1 and don't have any issues. 2.4.0 didn't work on 2023.2.X though.

binzhou, 2023-03-06:
P30-X and I'm just upgrade to 2.4.0 and 2.4.2 and still see the same error message.

slovenec88, 2023-03-07:
I had the same moving from 2.4.0 to 2.4.2. Remove all wallboxes, restart, then add back.

dannerph, 2023-03-11:
should be fixed in pre-releas 3.0.0, please check and confirm.

closed 2023-03-11 by dannerph

deqw, 2023-11-10:
@dannerph  I noticed something similar today:

> 2023-11-10 15:45:45.311 ERROR (MainThread) [homeassistant.components.notify] Error setting up platform keba
Traceback (most recent call last):
 
 File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 89, in async_setup_platform
    notify_service = await platform.async_get_service(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/notify.py", line 31, in async_get_service
    return KebaNotificationService(keba)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/notify.py", line 39, in __init__
    self.targets: dict[str, Wallbox] = {}
    ^^^^^^^^^^^^
AttributeError: property 'targets' of 'KebaNotificationService' object has no setter

Also 2 warnings:

> 2023-11-10 15:45:45.037 WARNING (MainThread) [homeassistant.components.number] custom_components.keba.number::KebaNumber is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author

> 2023-11-10 15:45:47.033 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.keba_p30_maximum_current_system_2 (<class 'custom_components.keba.sensor.KebaSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['mA', 'A']; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'keba' custom integration