dannerph/homeassistant-keba #9: Error when restarting while using multiple P30

richardpolzer, 2023-03-08:
When using this integration with multiple charging stations it works fine while setting up and afterward. Unfortunately on the restart of Home Assistant, the integration is only able to set up the first charging station. The second one becomes unresponsive. Interestingly the second wallbox is now in the integration whwrw only the first wallbox was. The entity all show not available though.

Attached I send some screenshots and debug outputs. Currently, I have 2 charging stations set up. There are 4 keba P30 installed here if that can be of help.

[image]
[image]
[image]

Logger: homeassistant.config_entries
Source: custom_components/keba/__init__.py:72
Integration: Keba Charging Station
First occurred: 10:25:09 (1 occurrences)
Last logged: 10:25:09

Error setting up entry P30 for keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/keba/__init__.py", line 72, in async_setup_entry
    wallbox = await keba.setup_wallbox(entry.data[CONF_HOST])
  File "/usr/local/lib/python3.10/site-packages/keba_kecontact/connection.py", line 144, in setup_wallbox
    self._wallbox_map[host] = self._wallbox_map.pop(wb.device_info.host)
KeyError: '192.168.xxx.x' _[redacted]_

Logger: homeassistant.components.notify
Source: custom_components/keba/notify.py:39
Integration: Benachrichtigung ([documentation](https://www.home-assistant.io/integrations/notify), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+notify%22))
First occurred: 10:25:10 (1 occurrences)
Last logged: 10:25:10

Error setting up platform keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 89, in async_setup_platform
    notify_service = await platform.async_get_service(
  File "/config/custom_components/keba/notify.py", line 31, in async_get_service
    return KebaNotificationService(keba)
  File "/config/custom_components/keba/notify.py", line 39, in __init__
    self.targets: dict[str, Wallbox] = {}
AttributeError: can't set attribute 'targets'

slovenec88, 2023-03-09:
I see the same behavior with P30X and P30C. If I remove them, and re-add them, they work perfectly until the next restart.  
I tried experimenting with "Enable newly added entities" system settings, before restarting, but it didn't help.

dannerph, 2023-03-09:
Thanks for reporting, was it working with an older version? I need to check that with my simulated wallboxes again.

slovenec88, 2023-03-09:
Yes. It was working pre 2.4.0 + pre HA 2023.2.X. Let me know if I can help somehow?

dannerph, 2023-03-11:
should be fixed in pre-releas 3.0.0, please check and confirm.

closed 2023-03-11 by dannerph

richardpolzer, 2023-03-12:
Thanks for the quick fix. It works even through restarts.