dannerph/homeassistant-keba #10: Entities unresponsive when renaming the charging station

richardpolzer, 2023-03-12:
When you rename the charging station and choose to rename entities, the entities become unresponsive until the next restart. Then they work as expected.

Is that expected behavior?

HA Version: 2023.3.3
Plugin Version: 3.0.0

dannerph, 2023-03-12:
Unfortunatelly, I am not able to reproduce this behaviour. With advanced mode enabled and choosing rename entities, all entities are renamed accordingly. The only thing I noticed is that the frontend does not update instantly but needs a refresh in my setup. in the backend (home assistant core) all works as exepcted.

richardpolzer, 2023-03-12:
You are right. My bad. It was just the UI not reloading. Thanks for your work.

closed 2023-03-12 by richardpolzer