dannerph/homeassistant-keba #11: Authentication Sensor not availible

richardpolzer, 2023-03-12:


dannerph, 2023-03-12:
I thought of only adding a lock sensor, if RFID is added to the configuration of the charging station. Without RFID configured, the lock is not working anyhow, at least with my BMW Wallbox Plus. Could you elaborate on how you use the lock?

richardpolzer, 2023-03-12:
Currently, I use the Lock to determine if the Charging Station (session) is authenticated or not. This allows me to only send the keba.start command to the charging stations that are locked and have an EV connected.

dannerph, 2023-03-12:
Good point, have not thought of that use case. I will re-add them tomorrow.

dannerph, 2023-03-12:
Quick fix: add a RFID to the configuration of the device. It should reload the component and add the missing locks.

richardpolzer, 2023-03-12:
Just as background info on the use case. Here we have an ekey xline Fingerprint sensor. That one sends a request to a webhook on HA. There I find the charging station that I just plugged in and authenticate that one with the RFID set by the finger that was used.

commit 12dd4a6e (2024-12-23): fix #11, #17, #19
closed 2024-12-23 by dannerph