dannerph/homeassistant-keba #13: service keba.stop does not work

mattsve, 2023-06-01:
Calling service keba.stop (no rfid specified) does nothing, the car keeps on charging.
The disable button works fine, and so does keba.start service after the disable button is pressed.

Tried with version 3.0.3

deqw, 2023-09-21:
I’m very happy with this integration because I now I’m no longer dependent on the (slow) BMW webservice to see if the car is charging or not. But like above I have a problem with starting and stopping charging via the integration. I have no RFID in use and therefore could not specify anything in the integrations setup. Start and stop now resulting in a authentication error. Is there a way to circumvent the RFID sequence?

matlimatli, 2023-11-26:
I changed my scripts/automations to use
```
service: button.press
data: {}
target:
  entity_id: button.keba_p30_enable
```
for starting, (and `button.keba_p30_disable` for stopping). Could this be an option for you as well?

pos-ei-don, 2023-11-27:
I do the same. As long as the charging is not stopped by the car (becaus the condifured soc is reached) enable will restart charging directly after hitting enable.
I the other case, when the car stopped the charging, I have to send the "charging" command to the car via api aswell.

deqw, 2023-11-27:
> I changed my scripts/automations to use
> 
> ```
> service: button.press
> data: {}
> target:
>   entity_id: button.keba_p30_enable
> ```
> 
> for starting, (and `button.keba_p30_disable` for stopping). Could this be an option for you as well?

That is helpful @matlimatli . I compared this to the call that the Call Service node is making in Node-Red:

[image]

So this is not resulting in the same call. I cannot get to a button.press service as far as I can see now. What I do see is that I can select a Keba domain in the node which can be used to start and stop. Which results in the following call.

[image]

I have to test if that is doing anything / the same as your automation call.

deqw, 2023-11-27:
Although the call is looking different it isn't. 

I see that:

[image]

Is picked up fine by the API:

[image]

The keba domain start/stop doesn't do anything from Node-Red.

matlimatli, 2023-11-28:
Actually, I have the same problem as OP. When I use the disable button, the charger sometimes suddenly starts charging again. I haven't really figured out why and when, but it is quite problematic for my charging scripts. So I would really like to be able to use the keba.stop service, which at least previously worked well.

dannerph, 2024-12-22:
@mattsve sorry for this very late response, I am currently updating the underlying library. Which version of the charging station do you have? If you do not have RFID, start and stop will do nothing. This is by purpose and should probably be renamed authoratization.

dannerph, 2024-12-23:
> Calling service keba.stop (no rfid specified) does nothing, the car keeps on charging. The disable button works fine, and so does keba.start service after the disable button is pressed.
> 
> Tried with version 3.0.3

I guess that is the intended behaviour (https://www.keba.com/download/x/4a925c4c61/kecontactp30udp_pgen.pdf)

The command stop can be used to deauthorize a charging session if the authorization function is activated. The effect is the same as holding an RFID card in front of the RFID reader during a running charging session. The **deauthorization will only work correctly if the RFID tag used is the same as the one used to start the charging session**.

closed 2024-12-23 by dannerph

Kolbi, 2025-02-11:
Why is there a start / stop service but no enable / disable?

https://github.com/dannerph/keba-kecontact/blob/master/keba_kecontact/charging_station.py

`    async def start(
        self, rfid: str | None = None, rfid_class: str = "01010400000000000000"
    ) -> None:
        """Authorize a charging process with given RFID tag.

        Args:
            rfid (str | None, optional): RFID tag to authorize. Defaults to None.
            rfid_class (str, optional): RFID class/color. Defaults to "01010400000000000000".

        """
        if KebaService.START not in self.device_info.services:
            raise NotImplementedError("Start is not available for the given charging station")

        cmd = "start"
        if rfid is not None:
            validate_rfid_tag(rfid)
            validate_rfid_class(rfid_class)
            cmd = f"start {rfid} {rfid_class}"

        await self.set_ena(True)
        await self._send(cmd, fast_polling=True, blocking_time_s=1)

    async def stop(self, rfid: str | None = None) -> None:
        """De-authorize a charging process with given RFID tag.

        Args:
            rfid (str | None, optional): RFID tag to authorize. Defaults to None.

        """
        if KebaService.STOP not in self.device_info.services:
            raise NotImplementedError("Stop is not available for the given charging station")

        cmd = "stop"
        if rfid is not None:
            validate_rfid_tag(rfid)
            cmd = f"stop {rfid}"

        await self._send(cmd, fast_polling=True, blocking_time_s=1)`

`    async def enable(self) -> None:
        """Start a charging process."""
        await self.set_ena(True)

    async def disable(self) -> None:
        """Stop a charging process."""
        await self.set_ena(False)`

dannerph, 2025-02-12:
@Kolbi 
In the past of home assistant, must actions required services to be implemented, especially when you need parameters, like an RFID tag. Later, they introduced buttons, which call an action but only without parameters. Thus, I implemented enable and disable as buttons because that is the common practise. See here:
https://github.com/dannerph/homeassistant-keba/blob/main/custom_components/keba/button.py#L34

I hope that answers your question? :)

pos-ei-don, 2025-02-13:
So make a Template switch, which checks the Status and pushes the enable/disable Button when needed.

Kolbi, 2025-02-13:
> So make a Template switch, which checks the Status and pushes the enable/disable Button when needed.

That is excatly what I'm doing curently. Using `{{ is_state('binary_sensor.keba_p30_keba_p30_charging', 'on') }}` as state.