dannerph/homeassistant-keba #14: KEBA station slutade fungera efter HA uppdateringen till 23.05

ELFHOUSE, 2023-06-07:
hejsan min box slutade fungera i HA efter maj uppdateringen detta kommer upp i loggen:
2023-06-07 18:48:28.613 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration keba which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2023-06-07 18:48:58.715 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry P30 for keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/__init__.py", line 139, in async_setup_entry
    hass.config_entries.async_setup_platforms(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'
2023-06-07 18:48:59.572 ERROR (MainThread) [homeassistant.components.notify] Error setting up platform keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 89, in async_setup_platform
    notify_service = await platform.async_get_service(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/notify.py", line 31, in async_get_service
    return KebaNotificationService(keba)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/notify.py", line 39, in __init__
    self.targets: dict[str, Wallbox] = {}

dannerph, 2023-11-15:
which version? still working for me.

ELFHOUSE, 2023-11-18:
Sins it been several update it is now working after I hade remove it and reinstalled it in ha 2023:7

closed 2023-11-19 by dannerph