dannerph/homeassistant-keba #17: Set current gets reset

PulsarFX, 2023-06-10:
If I change the value of "set current" it stays like this until the next charging session starts (or until it is unplugged, dunno)
Is this intended? 
Where does this value come from, as it resets to 63A. The slider max value is 32A. 
On the Ladeverbund Einstellungen, 48A is the highest value that exists.

spali, 2023-10-29:
Read the keba udp documentation, it's the difference between the commands curr and currtime

matlimatli, 2023-12-02:
It might also be worth noting that the underlying lib (keba_kecontact) have changed from using `curr` to `currtime` for the `set_current` service. This means that previously (in version 2.4.0 and earlier) the current setting was permanent, while now it is only temporary.

pos-ei-don, 2023-12-09:
I noticed this aswell. Xcan we change this back to the old behaviour for this plugin?

dannerph, 2024-12-22:
Hi all, as "curr" is permanent, it is writing to the internal chip. To avoid damaging the chip due to dozens of permanent write, I changed the behaviour by using "currtime". So this behaviour is basically intended.

The slider uses the "Curr HW" set in the charging station from report 2, which is the "Maximum current value in mA that can be supported by the hardware of the device. This value represents the minimum of the DIP switch settings, cable coding and temperature monitoring function."

I think a good compromise is to use "curr" for the slider and "currtime" for the service function, which is most likely integrated into automations and called dozens of times?

spali, 2024-12-23:
Didn't know curr is saved in a potentially limited write store.
In this case I'm not even sure if you should expose it. Or maybe only via a by default disabled entity.

For my use case it doesn't matter if the value is temporary, as i control it via helper entity, and is set everytime a new charging session starts.

closed 2024-12-23 by dannerph