dannerph/homeassistant-keba #20: Authorization rejected

deqw, 2023-10-11:
When plugging in the cable (P30) charging starts and then stops right away with Status: authorization rejected. No RFID installed or activated. All firmware and software with latest versions as of today. Did already a P30 factory reset to no avail and also tried with disabled integration. No automations running.

Diagnostic: AuthON: Off, Session en reason: 5, State: 5

Tried with a granny charger which works fine.

P30 v 3.10.36 (211117-093932)

Anyone here with a solution? Thanks!

deqw, 2023-11-27:
Still struggling with this, but making some progress, I think.

- In the integration configuration settings I have set now an 8 number UID for the RFID tag for default authentication.
- Then I have added a RFID Card to the Keba whitelist with the same UID (go `whitelist.php` on your local Keba URL).

In the Keba Device configuration I have set:
- Authorization: On
- Online Authorization Mode: OnlyLocal
- Offline Authorization Mode: OfflineNoAuthorization

The Keba timeout now is faster, when sending Enable, resulting in a faster updated Logbook, so the Keba is actually checking something locally now and not timing out against the OCPP. However `KEBA P30 Status  changed to authorization rejected` is still the result.

Using SolarEdge EV Charger (Keba P30 x-series), Firmware (SolarEdge): 1.14, Keba Integration v2.4.0.

Any tips would be appreciated.

deqw, 2023-11-28:
Got the 4.19.36 update today [https://www.solaredge.com/en/support/setapp-inverters-firmware#Released32111](url).
Did some testing and was able to charge now, but don't see that the RFID Card actually is used in the session; the RFID number is not in the logged session. Charging could be stopped too but then started again after a few seconds. Had to remove the cable from the car to stop charging.

deqw, 2023-12-04:
@dannerph 
When debug logging I see:
"RFID tag": "0000000000000000",
"RFID class": "00000000000000000000"

Where do these zero's come from? call.data / entry.options?
These are always in the debug data, also when I add an RFID card to the charger or add an RFID tag for default authentication in the Options eg. using 00845500.

BTW: once RFID tag/class configured in the Options I can remove and then submit these options but the once set values persist.

dannerph, 2024-01-13:
> @dannerph When debug logging I see: "RFID tag": "0000000000000000", "RFID class": "00000000000000000000"
>
> Where do these zero's come from? call.data / entry.options? These are always in the debug [...]

Those balues in debug should be just what the charging station sent an what the integration received. Not sure if those RFID values are handled consistantly among all types of boxes

closed 2024-02-26 by deqw