dannerph/homeassistant-keba #22: If secondary Wallbox is configured and not reachable, the first stopps working

pos-ei-don, 2023-11-01:
If secondary Wallbox is configured and not reachable, the first stopps working after restart of HA.
The Error is "error on initialize".
The Entities are there, but changing them stopps working. Even if I deactivate the second Wallbox.

I expected the first wallbox to work normal. The second lost ists power due to not enough solar-power, so this is daily 
happening here.

pos-ei-don, 2023-11-09:
```
Logger: homeassistant.components.number
Source: components/number/__init__.py:200
Integration: Nummer ([documentation](https://www.home-assistant.io/integrations/number), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+number%22))
First occurred: 12:11:17 (1 occurrences)
Last logged: 12:11:17

custom_components.keba.number::KebaNumber is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author
```

pos-ei-don, 2023-11-09:
Changed the Subject because  the problem always happens if two wallboxes are configured.

Adding a second works until next restart. My current workaround is to manually configure both wallboxes after every restart.

pos-ei-don, 2023-11-27:
Here a Screenshot how it looks like after EVERY restart of HA. (having the logs above in my systemlog)
If I delete both entitys and manually readd them, I can use them.

[image]

PulsarFX, 2023-11-28:
As far as I know, the problem is, that the integration puts both boxes to the same integration entry, because of ... (unclear as I struggle with it, too)

It says: 2 Geräte/devices. This always causes trouble on my side.

What I do is: 
- add first box. 
- rename it in the device dialog 
   - Make sure you have enabled "advanced mode" in your profile settings, so the rename dialog will ask you to rename all entities, too. Without this, renaming is useless.
   - click "1 Gerät" and 
   - click the pencil on the upper right corner. 
   - rename it and also let it rename all entities. 

- add second box and rename it as well for easier recognition.

renaming from the device list (shown in the screenshot) does not help on this problem.

pos-ei-don, 2023-11-28:
This makes a lot of troubles using automation and Templates. I cannot redo everything after each restart without errors. 
I hope, someone can fix this.

pos-ei-don, 2023-11-29:
Yes, it is fixed there! Thank you a lot for this hint!

closed 2023-11-29 by pos-ei-don