dannerph/homeassistant-keba #24: Setup issues with multiple boxes

PulsarFX, 2023-11-28:
When adding multiple wallboxes, the integration gets confused because the devices all have the same name.

This is shown in the integration entry list when "2 Geräte/devices" is shown. After a reboot, you get error messages that the integration can't be set up:

[image]

What I have to do, to get into a state, where reboots are fine and identifying entities in a save way is the following:
- add 1st box. 
-  rename it (whit advanced mode enabled in user profile settings) from the device list (pencil in the upper right corner)
- let HA rename the entities, too. This will include the new name in the entity names.
- add 2nd box.
- ...

This is cumbersome and error prone. If you mess it up in the first try, you will never be happy with this install again. I removed all boxes from HA, rebooted, removed all "problems" in developer settings > statistics.
re-added the boxes like above, but after rebooting HA, the issue comes back. 

Q: can the setup assistant use a custom name right away, so this rename procedure is not needed?

Q: is multicast part of the setup after a restart? 
Q: may be a eichrechtskonforme x-Wallbox a problem? 

I have 2 charging networks with 5 boxes each. One is fine, the other one makes trouble on HA restart. :-(

PulsarFX, 2023-11-28:
Ok, maybe the issue is starting earlier.
After the first box added, I get this in the logs:
```
Logger: homeassistant.components.notify
Source: components/notify/legacy.py:89
Integration: Benachrichtigung (documentation, issues)
First occurred: 21:07:38 (1 occurrences)
Last logged: 21:07:38

Error setting up platform keba
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 89, in async_setup_platform
    notify_service = await platform.async_get_service(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/notify.py", line 31, in async_get_service
    return KebaNotificationService(keba)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/keba/notify.py", line 39, in __init__
    self.targets: dict[str, Wallbox] = {}
    ^^^^^^^^^^^^
AttributeError: property 'targets' of 'KebaNotificationService' object has no setter
```

This entry is repeated after renaming all entities via described way in 1st post. 

Furthermore I find this warning in the logs:

```
Logger: homeassistant.core
Source: core.py:1903
First occurred: 21:12:51 (2 occurrences)
Last logged: 21:12:51

Unable to remove unknown service notify/keba_p30
Unable to remove unknown service notify/keba
```

dannerph, 2023-11-28:
Hi @PulsarFX 
yeah currently it seems there is an isssue with the notification. My only charging station does not have a display thus i most probably missed this issue.
I am a bit out of free time but hope to be able to improve the integration during xmas break.

PulsarFX, 2023-11-28:
I just discovered, that there are a ton of pre-releases available. I installed 3.0.3 and all issues are gone (including the notify)!! I take it ;-D

closed 2023-11-28 by PulsarFX

deqw, 2023-11-28:
Yup, waiting for a binary release too.