dannerph/homeassistant-keba #26: How do I get this great Keba integration to work?

User-Roman, 2023-12-31:
Hello Philipp,
I have read a lot about your Keba project here and find it very exciting.
I have only had HA installed for a few weeks and am still a real newbee.
At first I had Keba running via the "configuration.yaml" entry (I don't think this has anything to do with your project).
Then I managed to find the right link from GitHub (real challenge for a newbee) to include it under HACS/"Custom Resposertories" and download it afterwards by this. After the restart I got the message that I should remove the previously created Keba entry (still from the first attempt of a Keba integration) in the configuration.yaml. After the removal (and restart), the entities under "Current entities" of the first Keba integration are no longer available (unavailable), but no new ones have been added. 

Have I done something wrong or do I need to do something else? 

If you can and want to help me, detailed instructions/executions would be much appreciated (newbee, I'm still struggling with almost every instruction I'm supposed to execute).
I'm also still not sure what "Python library for controlling KEBA charging stations" is all about. Apart from that, what I would have to do to integrate it,  I think it is a part from your project.

Greetings, Roman

dannerph, 2024-01-13:
Hi @User-Roman 

Sorry for my late reesponse. You should remove the configuration from the config file. After you installed the custom integration (Google for HACS if you want to enhance your system further) and restartet home assistant, you should be able to install keba similat to this guide:
https://www.home-assistant.io/getting-started/integration/

The sensor names have changed, so it will most likely be necassary to update your dashboard afterwards.

Greetings Philipp

User-Roman, 2024-01-17:
Hello Philipp, 
Thank you very much for your reply! 
I have removed the Keba configuration from the yaml-config file and removed my Keba integration installed in HACS to eliminate any problems. Then I installed the Keba integration again  (beta 3.0.3) in HACS.
Unfortunately, when I search for entities named keba, I get no values. 

What am I doing wrong?

BR Roman

[image]

dannerph, 2024-01-17:
Hi @User-Roman 
have you set the dip switches correctly to enable the udp interface of your charging station?
To get the right dip settings, please check the manual of your charging station

#27

User-Roman, 2024-01-25:
Hi dannerph, I believe that I have the right settings or am I wrong?

[image]

dannerph, 2024-01-25:
@User-Roman could it be, that the values you showd in the screenshot above with unavailable are not from the new integration? Please check the device page unter integrations (Geräte und Dienste/Keba Charching Station/Gerät

User-Roman, 2024-01-26:
Hi dannerph, great thanks to you! That was the problem.  After I reinstalled the Keba integration I get the values displayed. Thanks again 😀

[image]

[image]

closed 2024-06-23 by dannerph