dannerph/homeassistant-keba #27: Lack of support for the Keba Wallbox P30

TSK7M, 2024-01-03:
Hello, 
unfortunately a lot of data is not evaluated, can the function be extended to include the missing data?
It would be nice if the features were available. 
Unfortunately, it doesn't work for many. As the plugin probably needs to be updated. 

KEBA P30
Firmware: P30 v 3.10.53 (230713-211537)

[image]

Thank you and greetings

pos-ei-don, 2024-01-03:
Does the p30 Website itself include this data? Do you have a p30 Model which Hasel this data?
My p30 Model does not Provider this Info!

TSK7M, 2024-01-07:
@pos-ei-don I can see this data on WebIF port 8080 --> Of course, I don't know what interface you're using.

[image]

[image]

TSK7M, 2024-01-09:
KEBA P30 Total energy --> sensor.keba_p30_total_energy

Unfortunately, the entity ID "sensor.keba_p30_total_energy" does not contain any data,
but it would be very interesting to determine the consumption in Home Assistant. How can the value be read?

[image]

gerione, 2024-01-12:
Which version of the P30 are you using? I am using the P30X and my sensor is populatd. 
[image]
[image]

TSK7M, 2024-01-12:
Thanks for your support.
I have the Kebea:
KC-P30-ES2400E2-L0A-

TSK7M, 2024-01-12:
KeContact P30 x-series
[image]

gerione, 2024-01-12:
So we have the same. Thats really interessting.

dannerph, 2024-01-12:
Hi @TSK7M 
could you please check again if the DIP switches are correctly set to enable UDP (check your installation manual)?
If I remember correctly, some values, like the session energy, are broadcasted by the charging station and thus interpreted by the home assistant integration. Other values need to be requested first. If the UDP interface is not enabled on the charging station, those requests are ignored.

If that is not working, please provide debug logs
```
logger:
  default: warning
  logs:
    homeassistant.components.keba: debug
    # wrong: keba: debug
    keba_kecontact: debug
```

TSK7M, 2024-01-12:
Hi @dannerph,
thank you! It was the DSW setting. 
After adjustment, all data was displayed.

----------

[image]

closed 2024-01-12 by TSK7M

TSK7M, 2024-01-14:
This is a good idea or you should see a message that the configuration is not correct...