dannerph/homeassistant-keba #32: Get RFID tag

nils512, 2024-09-11:
I'm using Home-Assistant 2024.9.1.
I got the Keba P30x with the RFID interface.

I like the feature, that i can add the RFID tag to Home-Assistant.
But how can i get the tag number?

I read the card with an NFC scanner on my phone and got the ID, ATQA and SAK. Do i need one of these values?
These hex-values were not working.

dannerph, 2024-09-11:
Hi @nils512 
of I remember correctly the hex values you read via NFC are big endian and the charging station requires the hex rfid code little endian. Thus if you convert the hex code to little endian it should work 🙂
Example: 01 02 03 (big endian) converts to 03 02 01 (little endian)

nils512, 2024-09-12:
ok thank you.

I used the ID if the card and convert it to little endian: _00:00:7E:7E:7E:7E:7E_ (first two values are 0, i changed the rest, to post it here)

I get this error message, when trying to unlock:
_RFID tag must be a 8 byte hex string._

Is the ID, really the correct value?

dannerph, 2024-09-12:
Ah I see, my cards only had 8 hex values, others seem to have more. I already identified the problem, which is really easy to fix.
Unfortunatelly I am on a business trip right now, thus the fix will be delayed. Please remind me in two weeks if I have not fixed it by then!

nils512, 2024-09-12:
no worries, thanks for checking the code :-)

I removed the colons so the value is: "00007E7E7E7E7E"
Home-Assistant is sending the value to the wall box, but the value is not accepted by the box.

Is the card id really the correct value?

Sorry, i have no experience with RFID/NFC tags.

nils512, 2024-09-30:
@dannerph : hey, do you have any news? Would be awesome to add the card to Home-Assistant :-)

I'm not sure, how to get the correct tag number from my card.

nils512, 2024-10-13:
i fixed it today :-)

For everyone else, that have no access to the webinterface, but to the USB port inside the wallbox:

1. get a FAT32 usb drive (other formats might also work)
2. open the wallbox, so that you have access to the usb port
3. Scan the card with the wallbox that you want to add to Home-Assistant
4. connect the USB drive to the wallbox
5. creating the lofgile takes about a minute. the display shows you the status
6. remove the usb drive
7. open the file located at "LOGS/P30_xxxxxxxx/P30_xxxxxxxx_diagnostics.tar.gz/kemove.log"
8. search for the log-entry: 
`[nioEventLoopGroup-3-1]  INFO (AuthorizationHandler:XX channelRead0) - AuthorizationReq [XXXXXXXX], principal: **00007E7E7E7E7E**, classifier , type RFID`
9. 00007E7E7E7E7E is the ID of the card. This value can be added to Home Assistant

closed 2024-10-15 by nils512