dannerph/homeassistant-keba #36: Key Error: 'host' in using KEBA services

schmersgahoven, 2025-01-08:
Hi all, 
got the following. After setting up Integration everything works fine for a couple of days or weeks. After some restarts if I want to call one of the KEBA services something went wrong.
[image]
and here are the logs to the specific failurte:
[image]

Sensor data and using sliders from the Integration: "KEBA P30 KEBA P30 Charging current" is working normally. 
Hope you have an idea and can help me.

schmersgahoven, 2025-01-08:
Hi,
I have manually entered the IP adress and it is only one charger. 
At the moment I am not using another custom integration for the KEBA for other stuff as well. But in the past in the past I tested with, KEBA Integration in Home Assistant, https://www.home-assistant.io/integrations/keba/
and was trying with evcc. But deleted evcc in the past and for the KEBA integration in HA I deleted the lines in the Konfiguration.yaml. 
But maybe there are some old trash data in the background which ist doing the behaviour that you mean with the device_id is not unique. Any idea how I can find and delete old stuff if there is some? Or delete link between KEBA and stored device_id and create a new one?

dannerph, 2025-01-08:
Yes, there could be an old entry stemming from the old setup (configuration.yaml), I would try the following:

1. Remove all keba integrations from the web configuration overview
2. Remove the keba integration from the configuration.yaml
3. Restart home assistant
4. Re-add the charging station with the web configuration
5. (Report back if the error persists.)

Theoretically the keba custom integration (beta) should recognize existing keba charging stations from the configuration.yaml and migrate them to the new setup, however, I have not tested that in much detail yet.

You can also search in the file `config/.storage/core.config_entries` (might be hidden, because one should not manipulate it) for the domain 'keba' if there is an entry with `"data":{}` then there is some old configuration still alive.

schmersgahoven, 2025-01-10:
Thanks for quick reply. Yesterday I tested your steps twice but sadly the error persists.
[image]

Had a look in the config/.storage/core.config_entries but there was no domain entry for 'keba'. By further investigation I was looking into file config/.storage/core.device_registry and there was still an entry of the keba after removing it from the web, in the configuration.yaml was no entry any more. So I deleted the line with 'keba' in the config/.storage/core.device_registry an re-add the KEBA with web configuration flow but error also still persists.
Do you have any further ideas?

dannerph, 2025-01-10:
It seems it there is a connection with the problem in #37 where the same error appears in the logs.

dannerph, 2025-01-13:
Which keba charging station do you have? Could you provide the producit Id? (KC-P30-XXXXXXXX-000)

schmersgahoven, 2025-01-14:
For sure: KC-P30-EC2401C2-EOR

schmersgahoven, 2025-03-20:
Hi all,.
any new ideas? Problem still persits.

benjwin, 2025-07-30:
I had the same error and pinned it down with some additional print-statements in `__init.py__`:
```
def _get_charging_station(
    hass: HomeAssistant, device_id: str
) -> ChargingStation | None:
    _LOGGER.error("####---TEMPORARYLOG calling _get_charging_station %s", device_id)
    # Get and check home assistant device linked to device_id
    device = device_registry.async_get(hass).async_get(device_id)
    if not device:
        _LOGGER.error("Could not find a device for id: %s", device_id)
        return None
    _LOGGER.error("####---TEMPORARYLOG device is %s", device)

    # Get and check config_entry of givne home assistant device
    config_entry = hass.config_entries.async_get_entry(
        device.primary_config_entry
    )
    if config_entry is None:
        _LOGGER.fatal("Config entry for device %s not valid", str(device))
        return None
    _LOGGER.error("benjdebug config_entry is %s", config_entry)

```

The output showed that two config entries are linked to my device: 

> 
>  ####---TEMPORARYLOG device is 
>         DeviceEntry(area_id=None, 
>             **config_entries={'01JMPMY5KXPN7GCRBMB1JVFR2X', '29063446c76ae6952207bee54628bbe2'},** 
>             config_entries_subentries={'01JMPMY5KXPN7GCRBMB1JVFR2X': {None}, '29063446c76ae6952207bee54628bbe2': {None}}, 
>             configuration_url='http://192.168.<<redacted>>', 
>             connections=set(), 
>             created_at=datetime.datetime(1970, 1, 1, 0, 0, tzinfo=datetime.timezone.utc), 
>             disabled_by=None, entry_type=None, hw_version=None, 
>             id='6e90f159069a65d7ead5635fcc0f6880', 
>             identifiers={('keba', '214<<redacted>>')}, labels=set(), 
>             manufacturer='KEBA', model='P30', model_id=None, 
>             modified_at=datetime.datetime(2025, 2, 22, 10, 58, 32, 703586, tzinfo=datetime.timezone.utc), 
>             name_by_user=None, name='KEBA P30', 
>             primary_config_entry='29063446c76ae6952207bee54628bbe2', 
>             serial_number=None, suggested_area=None, sw_version='P30 v 3.10.51 (230522-102136)', 
>             via_device_id=None, is_new=False, 
>             _cache={'json_repr': b'<<redacted>>'})

When looking up both config entries in the file _core.config_entries_, it shows that the 2906344.. is the Keba-Configuration (with the host correctly set) and the 01JMPM... is a template-helper (with no host-property), which I created via the gui:

> {
>     "created_at": "1970-01-01T00:00:00+00:00",
>     "data": {
>         "host": "192.168.<<redacted>>"
>     },
>     "disabled_by": null,
>     "discovery_keys": {},
>     "domain": "keba",
>     "entry_id": "29063446c76ae6952207bee54628bbe2",
>     "minor_version": 1,
>     "modified_at": "2025-05-10T06:08:58.658859+00:00",
>     "options": {
>         "failsafe": false,
>         "failsafe_fallback": 6,
>         "failsafe_persist": false,
>         "failsafe_timeout": 30,
>         "rfid": "",
>         "rfid_class": ""
>     },
>     "pref_disable_new_entities": false,
>     "pref_disable_polling": false,
>     "source": "import",
>     "subentries": [],
>     "title": "P30",
>     "unique_id": "214<<redacted>>",
>     "version": 1
> }

> 
> {
>     "created_at": "2025-02-22T10:58:32.701639+00:00",
>     "data": {},
>     "disabled_by": null,
>     "discovery_keys": {},
>     "domain": "template",
>     "entry_id": "01JMPMY5KXPN7GCRBMB1JVFR2X",
>     "minor_version": 1,
>     "modified_at": "2025-02-22T10:58:32.701651+00:00",
>     "options": {
>         "device_class": "power",
>         "device_id": "6e90f159069a65d7ead5635fcc0f6880",
>         "name": "keba_p30_charging_power_w",
>         "state": "{{ states(\"sensor.keba_p30_charging_power\") |float * 1000 }}",
>         "state_class": "measurement",
>         "template_type": "sensor",
>         "unit_of_measurement": "W"
>     },
>     "pref_disable_new_entities": false,
>     "pref_disable_polling": false,
>     "source": "user",
>     "subentries": [],
>     "title": "keba_p30_charging_power_w",
>     "unique_id": null,
>     "version": 1
> },

I created a pull-request which should fix this issue by not choosing the first config-entry in the list (which could be the wrong one) but uses the **primary_config** property of the device entry (which is hopefully always the correct one; but if this should not be set, it falls back to using the first one of the config_entries-List)

benjwin, 2025-07-31:
> [@benjwin](https://github.com/benjwin) thanks for tracing down this bug. I do not have a template senosr thus didn't have the issue. Have you tested you changes with multiple charging stations? (You can use the underlying library to emulate one). I will try to check your PR soon 🙂

Unfortunately I don't have multiple charging stations. If you can give me further hints on how to emulate one, I can check. 
However my expectation would be that in case of multiple charging stations, each one would have a different __device_id__ (which is the only parameter given to the modified function). Since the changes only modify on how to identify the correct linked __config_entry for a given device_id__, it should work for multiple charging stations as well.

commit 510188fd (2025-08-21): fix #36: use primary config entry of device (if set) instead of first one
commit db0ca507 (2025-08-21): Merge pull request #45 from benjwin/main

fix #36: use primary config entry of device (if set) instead of first entry