dannerph/homeassistant-keba #37: current set from 16 A to 6A and no change possible

mebi2024, 2025-01-09:
Since last Update to 4.3.0 my current (number.keba_p30_charging_current) is set to 6 ampere. Before is was 16 ampere. If I change the 6 to 16 in the dialog box it is reset after a frew seconds to 6. 
I also tried to change the value 6 to 16 in services.yaml:
/config/custom_components/keba/services.yaml
line 52: "default: 6" changed to "default: 16"
line 88: "default: 6" changed to "default: 16"
but there is no change after reload KEBA customization and restart of home assistant. How can I change back the current to 16 A?

dannerph, 2025-01-09:
Hi @mebi2024,

I assume you operate a Keba P30 charging station? Could you please provide the product string (KC-P30-XXXXXXXX-000) e.g. from the web interface of your charging station.

Could you confirm that the failsafe mode is disabled? (check the corresponding sensor in HA)

BR
Philipp

mebi2024, 2025-01-09:
Hi @dannerph ,
thank you for the quick response.

The product string is:  KC-P30-ES2400E2-L0R  (x-series)

failsafe mode is not really known to me, sorry (newbie in home assistant since Nov. 2024).  
binary_sensor.keba_p30_failsafe_mode = "sicher"

Dec. 31st until Jan. 4th failsafe was "unsicher".
Jan 4th until Jan 7th failsafe was "nicht verfügbar"
failsafe since Jan 7th "sicher"
if failsafe mode has to be changed, please let me know how.

Thank you
Martin

dannerph, 2025-01-09:
Ok, it seems failsafe mode is activate on your charging station. Depending on the failsafe settings, it will automatically revert your set current.
Go to developer menu to execute a service with the following settings:
[image]
Afterwards failsafe mode should be permanently off and setting current should be working again.
If it does not work, please reopen the issue

mebi2024, 2025-01-10:
Good morning,
during a car was loading yesterday evening I got the error message: 
"Die Aktion keba_set_failsafe konnte nicht ausgeführt werden. Unknown error" - see screenshot, but it is still the same this morning after restarting the KEBA and restarting home assistant. 
[image]

I also activated the Debug-Protokoll from home assistant KEBA integration - see attached file.
[home-assistant_keba_2025-01-10T06-33-20.196Z.log](https://github.com/user-attachments/files/18371707/home-assistant_keba_2025-01-10T06-33-20.196Z.log)

Regards,
Martin

dannerph, 2025-01-10:
Could you please enable `sensor.keba_p30_current_hardware`. This value is used for the slider as maximum value.

mebi2024, 2025-01-20:
Does someone have any ideas how to change 6A to 16A with a script?
My problem is, that customers are disappointed to load with 4kW instead fo 11kW and I have to change it back as soon as possible to 16A.
Does ist make sence to go to a higher HA version?

dannerph, 2025-01-20:
The issue with setting the failsafe mode in the developer menu is due to some yet unknown issue in registering the charhing station to home assistant. I could not reproduce it so far. Probably I need to extend debug logging to check what is going on in your installation. (I am blocked with other work this week).

Regarding the failsafe mode issue, is it possible to set it in the web ui of the charging station? I do not own a x-series but I know it has a lot more options there. If not you can run the the underlying python library of this integration which also has a CLI: https://github.com/dannerph/keba-kecontact
In the CLI mode pnce connect to the charging station, enter "failsafe 0 0 0" to disable it.

You only have one charging station or do you have multiples connected in master/slave? 🤔

Was it working with older vetsion of the Integration? If yes you could downgrade to v3.x. if doing so, always remove the charging station from the device menu an re-add them afterwards.

mebi2024, 2025-01-20:
Thank you again for the fast response!

I did not find "failsafe" mode in the web ui. Due to my OCCP connection to a charging provider I cannot use OCCP to the KEBA web connection (also used with the KEBA App on mobile). Maybe I can change the for testing mid of the week.

THanks for the CLI hint, but currently I don't know how to run this with python.

I have only one KEBA and no multiples (no master/slave)

I deinstalled KEBA integration and installed 3.0.3 and got the same error. I updated to 4.3.0 and still the same error. Maybe I have to try 3.0.1

Is it possible, that the HA version 2025.x causes the error?

dannerph, 2025-01-20:
In your case, it is mainly related to the failsafe mode switching back the current to 6A after a while. Changing settings in OCCP will probably not help.

I need to find out why the configuration is no stored correctly, unfortunately earliest next week.

schmersgahoven, 2025-01-21:
Just an idea if it's really urgent to deactivate failsafe. You can try via modbus integration.

Fist set up modbus in your configuration.yaml
example:

[image]

You neet at least one entity so I take Failsafe current. And change to your IP adress of charging station

restart home assistant and go to action and write register as described in the KEBA manual

[image]

[image]

[image]

Example for Failsafe Timeout:

[image]

Afterwards you have to delete added lines in configuration.yaml and restart Home Assistant. Otherwise maybe you will have problem with this integration because both will communicate with KEBA

closed 2025-03-20 by mebi2024