dannerph/homeassistant-keba #38: Multiple RFID-IDs

MikeLNo4, 2025-01-19:
Hi there.
Great project there, works like a charm straight out of the box. Thank you so much!

I understand this integration uses one standard RFID-tag from the configuration. I have 2 cars and must separate their charging data. I used to do this with multiple rfid-cars for authorization. With home assistant I am trying to do this via an actionable notification and user input on mobile device. But for that I need an action to authorize wallbox with rfid-id as a parameter to override central setting in configuration file. Would that be an option for further development or is there another way to make this work?

Thank you,
M.

dannerph, 2025-01-20:
Hi @MikeLNo4 

This is already possible. Check out the actions in the developer menu. If no RFID is given in the service call, the default one from the settings will be used. Otherwise, the provided one will be sent to the charging station.

[image]

MikeLNo4, 2025-01-20:
Interesting. Thank you. Definetely don't see that action in the dev section. I will need to check if I am on the latest version of this integration. Will check once I m am home.

M.

MikeLNo4, 2025-01-20:
Found it. Indeed I was not running the beta, my bad!I see the parameter for Keba.star now. Thank you very much!!

closed 2025-01-20 by MikeLNo4

nils512, 2026-04-07:
I'm using HA 2026.4.1 and cant see this option in the developer tools:
[image]

how can i update?