dannerph/homeassistant-keba #39: Support for min and max time values of the display notifications

Xlinx64, 2025-02-01:
Is it still possible to set a minimum and maximum time for the display notifications, like how it used to work with the legacy integration?

```
data:
  min_time: 4
  max_time: 10

It would be nice if it were possible to set the time with separate entities.

dannerph, 2025-02-03:
Hi @Xlinx64 unfortunatlly it is no longer possible to set these settings in the service call anymore, however, I could add it as configuration per charging station (device). Does this fit to your needs or do you need different times per sent message?

dannerph, 2025-02-03:
Ah, I just checked, extended data should be feasible as well. I need some free time to implement it ...

Xlinx64, 2025-02-03:
Hi!
I am not completely sure, how I can use it yet. Extended data would probably be the best.

I planned on writing some text for a certain amount of time, when I come home. The wallbox would automatically unlock in this time and start charging. I would send a second text with 1s maxtime, to delete the message, when a charger is plugged in.

dannerph, 2025-02-12:
you are right, there is no option for extended data. What about setting a default timeout per device in an OptionsFlow? And in case you want to clean the message earlier, just send an empty message a bit delayed?
I do not have a display on my charging station, what is shown there by default?