dannerph/homeassistant-keba #41: Entity is named always *.keba_p30_keba_p30_*

kuzco1, 2025-04-23:
Have installed this on my test-system.
All entities have the name .keba_p30_keba_p30_*, for example: 
- binary_sensor.keba_p30_keba_p30_plugged_on_ev
- sensor.keba_p30_keba_p30_status

dannerph, 2025-04-24:
Hi @kuzco1 Thanks for reporting. Have you done a clean install or was the same charging stations activated before using the keba integration of the core or another beta version?

kuzco1, 2025-04-25:
Hi, I was "playing" with both options (core HA component) and also this custom-component (on my test HA system), but never at the same time. 
So, I believe I have done a clean install, because I have deleted the entry in configuration.yaml and have also checked that all old entities are no longer visible in HA before installing this custom-component.

closed 2025-08-21 by dannerph