dannerph/homeassistant-keba #42: Request: Serialnumber and FW-Version

kuzco1, 2025-04-23:
Can you please add an entity for serialnumber and firmware-version ?
Thnaks.

dannerph, 2025-04-24:
HI @kuzco1 Thanks for reporting. This information is already available in the device info of the charging stations. Could you please explain, why you explicitly require to have a sensor for serial number and firmware-version? Sensors are commonly used for dynamic data only. Those are not dynamic at all, thus should not be sensors.

kuzco1, 2025-04-25:
There is not really a need to get the info of product, serial, uptime or firmware. It is just a wish to see these info in my dasboard.
However, in the meantime I have edited the core-HA component to see these infos. They are now visible as an attribute in the charging_power sensor.