dannerph/homeassistant-keba #49: max charging current is set to 63A for some reason if charger  is rebooted

agvg1, 2025-08-15:
My charger has a 16A fuse and the charger current range range is from 6 - 16A, but after a  restart of the charger it shows av value of 63A instead of 16A, this is a bit strange, a click on the refresh data button seems to fix it.

Should it be a autorefresh funktion in the case charger has been offline so the values is correct?

agvg1, 2025-08-15:
Same issue with the set current value

[image]

dannerph, 2025-08-21:
Hi @agvg1 
what do you mean with "restart of the charger"? Have you disconnected the charger from the power supply? If yes, when in the chart: at around 17:00 (at the increase) or at 11:00 (at the decrease)?

agvg1, 2025-08-21:
Yes, it was after a power loss for a moment at ca 18.00, and about 11.00 I was charging and one of my scripts complained about values out of range, the script adjusts amp on the charger so i dont use over 5 kw at the meter, a refresh data did correct it.

agvg1, 2025-08-22:
Going to do some testing with the new version to see what behavior it has when the charger is cold started after power loss.

dannerph, 2025-08-22:
> Going to do some testing with the new version to see what behavior it has when the charger is cold started after power loss.

Should be the same reaction 🙂

I guess your script is using the set_current action? This is a non-permanent setter. Thus after a reboot, the cutrent is reset to tge previous permanent setting. If you use the slider, the current is set as permanent.
The maximum of the slider is fetched from the charging stations limit.
I would guess that your permanent current is set to 63A as initial permanent setting of your charging station?

agvg1, 2025-08-22:
yes, im going to do some tests, but if it defaults to 63A in a very short period after booting then the integration could do a fetch data and correct this error.

agvg1, 2025-08-22:
And 63A here to

[image]

dannerph, 2025-08-22:
In that state, what is shown on the web interface of the charging station? (if there is one)

agvg1, 2025-08-22:
[image]

agvg1, 2025-08-22:
I dont  remember if the car was plugged in the first time, now it was plugged in but not charging. 
Can try unplugged if that makes a change
Any easy way to log communication between charger and HA?

agvg1, 2025-08-22:
Could try to wireshark it if thats give any usefull information

dannerph, 2025-08-22:
I quickly did the same with my charging station. I also get 63A. This 63 A is the value that is fetched from the charging station. In particular there are two values in the json that is fetched:
"Curr HW": 25000, (DIP limit)
"Curr user": 63000, (user limit set by the slider)

Thus, I think the integration is behaving exactly as it should: The slider respresents the user limit. This limit however, is reset to a default value of 63 A at the charging station once rebooted.

I do not think it makes sense to correct anything on home assistant side with this respect.

closed 2025-08-22 by dannerph

agvg1, 2025-08-22:
This value stayes correct

[image]

agvg1, 2025-08-22:
So would it be a idea to correct the two values with the real hW limit?