dannerph/homeassistant-keba #53: ***SolarEdge system | Keba P30x terminal ***

micheldr, 2025-09-18:
Hello everyone
I would like to be able to modulate the power of the three-phase terminal. Neither the installer nor SolarEdge want to help me.
The consumption/production meter is integrated into the Backup equipment.

After research,
It turns out I should put the “Domestic TCP Connection Supervision” parameter to ON (currently it is OFF)
But what are the parameters to set?

Thank you for your help and wish you a pleasant day.

dannerph, 2026-01-04:
This is not really an issue of the homeassistant keba integration, isn't it?

closed 2026-01-04 by dannerph

dannerph, 2026-01-04:
For general question, please use the community board, e.g. here: https://community.home-assistant.io/t/evse-keba-p30-integration/52571/142